gpiozero/gpiozero #1138: Button not working with gpiozero but working with gpiod

harshvardhanagrawal, 2024-04-04:
Operating system: Debian GNU/Linux 12 (bookworm)
Python version: 3.11.2
Pi model: Pi 5
GPIO Zero version: 2.0
Pin factory used: LGPIOFactory

Basic button functioning is not working with gpiozero module. I tried using the basic example given in https://github.com/gpiozero/gpiozero/blob/master/docs/examples/button_1.py

```from gpiozero import Button

button = Button(23)

while True:
    if button.is_pressed:
        print("Button is pressed")
    else:
        print("Button is not pressed")

```

Button Press is not being detected with the above code.

I am using a 4-pin button.
->Connections are okay as I am able to detect using gpiod module
->gpiozero installation is okay as I am able to turn on led using gpiozero.

bennuttall, 2024-04-04:
Do you get an error? Are you in a virtualenv? How did you install lgpio?

harshvardhanagrawal, 2024-04-04:
> Do you get an error? Are you in a virtualenv? How did you install lgpio?

1. I do not get any error,
‘’’button.is-pressed’’’ prints false
2. I am running it on the system-installed python environment.
3. I did not install it explicitly.

ValentinSvidrigaylov, 2024-04-09:
same issue here, same setup as yours, did you manage to fix it somehow?

harshvardhanagrawal, 2024-04-09:
> same issue here, same setup as yours, did you manage to fix it somehow?

Nah. I switched to gpiod for now.

ValentinSvidrigaylov, 2024-04-09:
Well, I'm glad that you found a temporary solution. I also thought about using gpiod until this issue is fixed, but with gpiod python library I get OS error 16, so I guess I'm stuck here for now. Have you encountered this error too? If so, how do I resolve it?

yume-tea, 2025-01-20:
I had the same issue on a Raspberry Pi 5. LED's work just fine, but input from buttons are not received with gpiozero. gpiod works for the same setup, so it's something within gpiozero.