gpiozero/gpiozero #1166: lgpio pin factory is broken on RPi5 since kernel 6.6.45

HiassofT, 2024-08-21:
Operating system: e.g. RPiOS Bookworm
Python version: e.g. 3.11.2
Pi model: e.g. Pi 5
GPIO Zero version: 2.0.1
Pin factory used: lgpio

A recent change in the RPi kernel switched the gpiochip number on RPi5 to 0 (see https://github.com/raspberrypi/linux/pull/6144) but gpiozero still uses a hardcoded chip number of 4 on RPi5 (see https://github.com/gpiozero/gpiozero/blob/master/gpiozero/pins/lgpio.py#L66) and thus fails to find the GPIO pin

```
root@raspberrypi:~# python3
Python 3.11.2 (main, May  2 2024, 11:59:08) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import gpiozero
>>> b = gpiozero.Button(4)
/usr/lib/python3/dist-packages/gpiozero/devices.py:295: PinFactoryFallback: Falling back from lgpio: 'can not open gpiochip'
  warnings.warn(
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 408, in pin
    pin = self.pins[info]
          ~~~~~~~~~^^^^^^
KeyError: PinInfo(number=7, name='GPIO4', names=frozenset({'J8:7', 'BCM4', 'GPIO4', 4, 'WPI7', '4', 'BOARD7'}), pull='', row=4, col=1, interfaces=frozenset({'', 'gpio', 'dpi', 'spi', 'uart', 'i2c'}))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 103, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 412, in __init__
    super().__init__(
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 417, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 162, in __init__
    super().__init__(
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 243, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/input_devices.py", line 79, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 549, in __init__
    pin = self.pin_factory.pin(pin)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 410, in pin
    pin = self.pin_class(self, info)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 101, in __init__
    GPIO.setup(self._number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])
RuntimeError: Cannot determine SOC peripheral base address
```

Pierrot66657, 2024-08-21:
Bonjour, du coup comment je peux faire pour contrôler mon GPIO sur Raspberry Pi5 via Python, script bash ou via le terminal?

Merci

SVDouble, 2024-08-23:
I used this monkey patch to work around the issue:
```python
import gpiozero.pins.lgpio
import lgpio

def __patched_init(self, chip=None):
    gpiozero.pins.lgpio.LGPIOFactory.__bases__[0].__init__(self)
    chip = 0
    self._handle = lgpio.gpiochip_open(chip)
    self._chip = chip
    self.pin_class = gpiozero.pins.lgpio.LGPIOPin

gpiozero.pins.lgpio.LGPIOFactory.__init__ = __patched_init
```

This forces GPIO Zero to use chip 0 on RPi5 and resolves the problem.

HiassofT, 2024-08-23:
Yes, something like this or just patching https://github.com/gpiozero/gpiozero/blob/master/gpiozero/pins/lgpio.py#L66 will work as a quick workaround on newer kernels but break stuff on older kernels so this isn't a solution.

Something like this https://github.com/waveform80/rpi-lgpio/commit/bb7ff0e2072eb15ad81c713201a2616e69c464a8 rpi-lgpio change needs to be done for gpiozero's lgpio usage, too

Pierrot66657, 2024-08-24:
Merci pour vos retours mais depuis la 6.6.47, tout est redevenu fonctionnel!

Milliways2, 2024-09-09:
`gpiozero` using the default lgpio fails on a Pi5 using Kernel 6.6.47+rpt-rpi-2712.

GPIO has moved to `gpiochip0` but `gpiozero` uses `gpiochip4`.

Although the LGPIOFactory accepts chip as a parameter it ignores the value passed in and replaces it with 0.

gpiozero later checks the Revision Code and for BCM2712 replaces with gpiochip4.

To continue support for older kernel on Pi5 it would be better to test for the presence of `/dev/gpiochip4`.

pi-lemaire, 2024-09-09:
Hi there,
The last update on the Raspbery Pi 5 changed the GPIO chip ID, breaking my small test.
The simple fix was to change the environment variable:
`export RPI_LGPIO_CHIP=0`

1CM69, 2024-09-09:
> Hi there, Indeed the last update on the Raspbery Pi 5 changed the GPIO chip ID, breaking my small flashing lights test script. The simple fix was to change the environment variable: `export RPI_LGPIO_CHIP=0`

I'm having this issue too, where exactly are you adding this fix too (more details needed for me the dummy) thanks

pi-lemaire, 2024-09-09:
in the shell, before you call your python script
you may also add it at the beginning of your python script, before importing the lgpio library:

`import os 
os.environ['RPI_LGPIO_CHIP']='0'`

at least this worked for me

L4wnmower, 2024-09-09:
> export RPI_LGPIO_CHIP=0

Issue does not seem fixable through this

L4wnmower, 2024-09-09:
> Something like this [waveform80/rpi-lgpio@bb7ff0e](https://github.com/waveform80/rpi-lgpio/commit/bb7ff0e2072eb15ad81c713201a2616e69c464a8) rpi-lgpio change needs to be done for gpiozero's lgpio usage, too

Do we know if this commit was ever merged in gpiozero or not? [Waveform80 commented on July 18th](url=https://github.com/raspberrypi/linux/pull/6144#issuecomment-2236590839) he was checking to include this

pelwell, 2024-09-11:
This ought to work as a minimally invasive patch:
```diff
diff --git a/gpiozero/pins/lgpio.py b/gpiozero/pins/lgpio.py
index 9fa8739..81e0298 100644
--- a/gpiozero/pins/lgpio.py
+++ b/gpiozero/pins/lgpio.py
@@ -63,7 +63,7 @@ class LGPIOFactory(LocalPiFactory):
     """
     def __init__(self, chip=None):
         super().__init__()
-        chip = 4 if (self._get_revision() & 0xff0) >> 4 == 0x17 else 0
+        chip = 4 if (self._get_revision() & 0xff0) >> 4 == 0x17 and os.path.exists('/dev/gpiochip4') else 0
         self._handle = lgpio.gpiochip_open(chip)
         self._chip = chip
         self.pin_class = LGPIOPin
```

XECDesign, 2024-09-11:
> This ought to work as a minimally invasive patch:
>
> ```diff
> diff --git a/gpiozero/pins/lgpio.py b/gpiozero/pins/lgpio.py
> index 9fa8739..81e0298 100644
> --- a/gpiozero/pins/lgpio.py
> +++ [...]

Should I pull that in to our package of it?

pelwell, 2024-09-11:
Or there's this...:
```
$ sudo ln -s gpiochip0 /dev/gpiochip4
```

pelwell, 2024-09-11:
There's an unofficial patch and a more general workaround about to go into our gpiozero package, so the pressure is off.

L4wnmower, 2024-09-11:
> Sorry for my tardiness on this -- I've been intending to copy my [fix from rpi-lgpio](https://github.com/waveform80/rpi-lgpio/commit/bb7ff0e2072eb15ad81c713201a2616e69c464a8) into gpiozero for this but I've been swamped with work things. Unfortunately I'm still swamped tonight (typing this in the middle of a meeting :) but I'll stick this on top of my list for tomorrow morning.

Dave, thank you for the effort.
Many thanks also to Phil for his help and to Mathias for catching the issue early on

usr-fritz, 2024-09-12:
I found this article, which is also helpful, to fix the problem:

[https://kofler.info/gpio-aerger-auf-dem-raspberry-pi-5/](url)

lurch, 2024-09-12:
> There's an unofficial patch and a more general workaround about to go into our gpiozero package, so the pressure is off.

This "hacky patch" and "symlink workaround" (@XECDesign 's words :wink: ) are now both available on Raspberry Pi OS with an `apt full-upgrade`.

ksubileau, 2024-11-02:
I'm trying to run a project based on gpiozero from a docker container. It worked fine on kernel 6.6.31, but now since upgrading to 6.6.51 it's broken.

```
docker run --device /dev/gpiochip4 --privileged myproject:latest
```

I came across this issue after some research, and it seems to match what is described. Indeed, on my host, the /dev/gpiochip4 path is now a symbolic link to /dev/gpiochip4
[image]

So the device mount set in the docker run command no longer allows to access to the gpio chip at /dev/gpiochip4 (because the symlink is resolved by the command itself). Additionally, since the /dev folder is a tmpfs mount inside the container, it is not possible to create a symlink from /dev/gpiochip4 to /dev/gpiochip0 at build time using the Dockerfile. I have to create this symbolic link at run time just before launching the application to fix the problem.

For this specific use case but also for greater general flexibility, wouldn't it be better to allow the user to explicitly configure the device path?

pelwell, 2024-11-02:
Can't you just use /dev/gpiochip0 instead?

ksubileau, 2024-11-02:
@pelwell I couldn't find any way to tell the gpiozero library to use /dev/gpiochip0 instead of /dev/gpiochip4. Am I missing something?

pelwell, 2024-11-02:
It looks like you've told Docker to allow access to device /dev/gpiochip4, but the device is now /dev/gpiochip0 - gpiochip4 is just a symlink as you've said. Try with  /dev/gpiochip0 on the docker command line, and hope that the symlink will work for lgpio.

ksubileau, 2024-11-02:
Yes, I did that, but then I still get an error that seems to indicate that gpiozero (lgpio) is still looking for the device at the path /dev/gpiochip4: 

```
$ docker run --device /dev/gpiochip0 --privileged myproject:latest
/usr/local/lib/python3.11/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from lgpio: 'can not open gpiochip'
  warnings.warn(
/usr/local/lib/python3.11/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from rpigpio: No module named 'RPi'
  warnings.warn(
/usr/local/lib/python3.11/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from pigpio: No module named 'pigpio'
  warnings.warn(
/usr/local/lib/python3.11/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from native: unable to open /dev/gpiomem or /dev/mem; upgrade your kernel or run as root
  warnings.warn(
2024-11-02 22:00:16,277 [1] CRITICAL radio-exporter: Uncaught exception
Traceback (most recent call last):
  File "/app/main.py", line 196, in <module>
    sys.exit(main())
             ^^^^^^
  File "/app/main.py", line 174, in main
    with Radio(868.95, isHighPower=True,  # verbose=True,
         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/RFM69/radio.py", line 74, in __init__
    self._init_gpio()
  File "/app/RFM69/radio.py", line 92, in _init_gpio
    self.intPinDevice = DigitalInputDevice(pin=self.intPin, pull_up=None, active_state=True)
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/gpiozero/input_devices.py", line 162, in __init__
    super().__init__(
  File "/usr/local/lib/python3.11/site-packages/gpiozero/mixins.py", line 243, in __init__
    super().__init__(*args, **kwargs)
  File "/usr/local/lib/python3.11/site-packages/gpiozero/input_devices.py", line 79, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/usr/local/lib/python3.11/site-packages/gpiozero/devices.py", line 544, in __init__
    super().__init__(pin_factory=pin_factory)
  File "/usr/local/lib/python3.11/site-packages/gpiozero/devices.py", line 245, in __init__
    Device.ensure_pin_factory()
  File "/usr/local/lib/python3.11/site-packages/gpiozero/devices.py", line 270, in ensure_pin_factory
    Device.pin_factory = Device._default_pin_factory()
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/site-packages/gpiozero/devices.py", line 302, in _default_pin_factory
    raise BadPinFactory('Unable to load any default pin factory!')
gpiozero.exc.BadPinFactory: Unable to load any default pin factory!

```
The workaround I found is to add the symlink into the entrypoint script, at runtime (`ln -s gpiochip0 /dev/gpiochip4`). But I think it would be cleaner to be able to directly give the right path to the gpiozero lib.

pelwell, 2024-11-02:
It wasn't clear from your "broken" comment that you had got it working.

pokulo, 2024-11-03:
@ksubileau I ran into the same problem. When I remember correctly , my solution was to switch the `gpiozero` backend to `rpi-gpio==0.7.1` simply by adding this to the python dependency list in `requirements.txt` of my project. And as the warnings on-top of the traceback suggest, `gpiozero` searches for other hardware-mappings as a fallback:
```
/usr/local/lib/python3.11/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from rpigpio: No module named 'RPi'
  warnings.warn(
/usr/local/lib/python3.11/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from pigpio: No module named 'pigpio'
  warnings.warn(
/usr/local/lib/python3.11/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from native: unable to open /dev/gpiomem or /dev/mem; upgrade your kernel or run as root
```
So I simply map all of the existing mappings in the docker call :` --device /dev/mem --device /dev/gpiomem --device /dev/gpiochip0 --device /dev/gpiochip1 --device /dev/gpiochip2 `. Maybe this helps?

bennuttall, 2024-11-03:
It would be better to explictly choose a pin factory by setting an environment variable:

```
GPIOZERO_PIN_FACTORY=lgpio
```

lurch, 2024-11-03:
I suspect that part of the problem / confusion might be that the `gpiozero` included in Raspberry Pi OS has been patched to look for `/dev/gpiochip0` (to match the kernel included in Raspberry Pi OS), but the version of `gpiozero` included in `pip` still looks for `/dev/gpiochip4` on Raspberry Pi 5? And so my assumption is that @ksubileau 's Docker container is using the `pip` version of GpioZero (which looks for `/dev/gpiochip4`), but the Docker container is running on a Pi 5 on top of the Raspberry Pi OS kernel (which provides `/dev/gpiochip0`) ?

bennuttall, 2024-11-03:
Yeah that would be it! We still need to do a release with that patch.

pelwell, 2024-11-03:
This is the patch (from my [comment](https://github.com/gpiozero/gpiozero/issues/1166#issuecomment-2343325731) above):
```diff
diff --git a/gpiozero/pins/lgpio.py b/gpiozero/pins/lgpio.py
index 9fa8739..81e0298 100644
--- a/gpiozero/pins/lgpio.py
+++ b/gpiozero/pins/lgpio.py
@@ -63,7 +63,7 @@ class LGPIOFactory(LocalPiFactory):
     """
     def __init__(self, chip=None):
         super().__init__()
-        chip = 4 if (self._get_revision() & 0xff0) >> 4 == 0x17 else 0
+        chip = 4 if (self._get_revision() & 0xff0) >> 4 == 0x17 and os.path.exists('/dev/gpiochip4') else 0
         self._handle = lgpio.gpiochip_open(chip)
         self._chip = chip
         self.pin_class = LGPIOPin```

ksubileau, 2024-11-03:
Indeed @pokulo, this is probably another possible workaround (even if in my case /dev/gpiomem does not exist - only /dev/gpiomem0, 1, ...), but sorry as @pelwell noticed, I wasn't very clear initially, I found a solution that works for my case, and my motivatio was rather to show a use case among others where it would be interesting to be able to explicitly configure the device path rather than having it hardcoded

yunhao-jiang, 2025-02-20:
> I used this monkey patch to work around the issue:
> 
> import gpiozero.pins.lgpio
> import lgpio
> 
> def __patched_init(self, chip=None):
>     gpiozero.pins.lgpio.LGPIOFactory.__bases__[0].__init__(self)
>     chip = 0
>     self._handle = lgpio.gpiochip_open(chip)
>     self._chip = chip
>     self.pin_class = gpiozero.pins.lgpio.LGPIOPin
> 
> gpiozero.pins.lgpio.LGPIOFactory.__init__ = __patched_init
> This forces GPIO Zero to use chip 0 on RPi5 and resolves the problem.

^^ This works! Thank you so much!

petekubiak, 2025-11-21:
There's also the issue of #1162 which means that even if you want to work around this bug by creating your own LGPIOFactory with `chip=0` it just ignores the argument and fails anyway