gpiozero/gpiozero #1169: Broken support for PWM in LEDCharDisplay

etam, 2024-09-01:
Operating system: Raspbian Bookworm
Python version: 3.11.2
Pi model: 3 Model B Rev 1.2
GPIO Zero version: 2.0
Pin factory used: lgpio

The code:
```python
from gpiozero import LEDCharDisplay

pin1  = "BOARD15"
pin2  = "BOARD19"
pin3  = "BOARD21"
pin4  = "BOARD23"
pin5  = "BOARD27"
pin6  = "BOARD28"
pin7  = "BOARD26"
pin8  = "BOARD24"
pin9  = "BOARD22"
pin10 = "BOARD18"
c = LEDCharDisplay(pin3, pin9, pin8, pin6, pin7, pin4, pin1, dp=pin2, pwm=True)
```

Fails with
```
Traceback (most recent call last):
  File "/home/etam/./rpi_digits.py", line 13, in <module>
    c = LEDCharDisplay(pin3, pin9, pin8, pin6, pin7, pin4, pin1, dp=pin2, pwm=True)
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 103, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/boards.py", line 967, in __init__
    super().__init__(
  File "/usr/lib/python3/dist-packages/gpiozero/boards.py", line 300, in __init__
    **{
      ^
  File "/usr/lib/python3/dist-packages/gpiozero/boards.py", line 303, in <dictcomp>
    LEDClass(
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 103, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/gpiozero/output_devices.py", line 390, in __init__
    if not 0 <= initial_value <= 1:
           ^^^^^^^^^^^^^^^^^^^^^^^
TypeError: '<=' not supported between instances of 'int' and 'NoneType'
```

It's also not clear how it would be controlled. LEDCharDisplay "value" takes a char to display. The brightness value must be provided some other way.

bennuttall, 2024-09-02:
Hmm yes that seems like a bug. `LEDCharDisplay` constructs an `LEDCollection` with `initial_value=None` [here](https://github.com/gpiozero/gpiozero/blob/master/gpiozero/boards.py#L972) and then sets the value to the required initial value, which is a space by default. But `LEDCollection` requires a number between 0 and 1.

I think changing this from `None` to `0` should fix it.

Try using this version instead:

```python
from gpiozero import LEDCollection

class LEDCharDisplay(LEDCollection):
    def __init__(self, *pins, dp=None, font=None, pwm=False, active_high=True,
                 initial_value=" ", pin_factory=None):
        if not 1 < len(pins) <= 26:
            raise PinInvalidPin(
                'Must have between 2 and 26 LEDs in LEDCharDisplay')
        for pin in pins:
            if isinstance(pin, LEDCollection):
                raise PinInvalidPin(
                    'Cannot use LEDCollection in LEDCharDisplay')

        if font is None:
            if len(pins) == 7:
                with resources.files('gpiozero.fonts').joinpath('7seg.txt').open() as f:
                    font = load_font_7seg(f)
            elif len(pins) == 14:
                with resources.files('gpiozero.fonts').joinpath('14seg.txt').open() as f:
                    font = load_font_14seg(f)
            else:
                # Construct a default dict containing a definition for " "
                font = {" ": (0,) * len(pins)}
        self._font = LEDCharFont(font)

        pins = {chr(ord('a') + i): pin for i, pin in enumerate(pins)}
        order = sorted(pins.keys())
        if dp is not None:
            pins['dp'] = dp
            order.append('dp')
        super().__init__(
            pwm=pwm, active_high=active_high, initial_value=0,
            _order=order, pin_factory=pin_factory, **pins)
        if initial_value is not None:
            self.value = initial_value
```

lurch, 2024-09-02:
> It's also not clear how it would be controlled. LEDCharDisplay "value" takes a char to display. The brightness value must be provided some other way.

Yeah, if you're constructing an `LEDCharDisplay` with `pwm=True` there's currently no way of setting the pwm-values of the LEDs, without iterating through each individual LED in the collection. (and this pwm-value would then get overwritten whenever you turned on or off any LED)
Perhaps `LEDCollection` needs a sibling `PWMLEDCollection` class which has a `brightness` property or something, and then when you turn on an LED it would only set it to the `brightness` value (not `1.0`); and when you change the `brightness` value, any LED which isn't currently off would get updated to a value of `brightness` ? :thinking:

bennuttall, 2024-09-02:
Ah I see, that's why this is happening / why the tests aren't catching it.

And yes brightness setting would make sense.