gpiozero/gpiozero #1172: RuntimeError: Cannot determine SOC peripheral base address

dshaurya, 2024-09-21:
Operating system: 12 (bookworm) 64bit
Python version: 3.11.2
Pi model: Pi 5
GPIO Zero version: 2.0.1
Pin factory used: RPiGPIO
See http://rpf.io/gpzissue for information on how to find out these details.

When i am running the below code i am getting the following r

```python

from gpiozero import LED
from time import sleep

led = LED(18)

while True:
    led.on()
    sleep(1)
    led.off()
    sleep(1)

```
```
/home/pi/python_projects/myproject1/lib/python3.11/site-packages/gpiozero/devices.py:300: PinFactoryFallback: Falling back from lgpio: No module named 'lgpio'
  warnings.warn(
Traceback (most recent call last):
  File "/home/pi/python_projects/myproject1/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 411, in pin
    pin = self.pins[info]
KeyError: PinInfo(number=12, name='GPIO18', names=frozenset({'BOARD12', 18, 'WPI1', 'BCM18', '18', 'J8:12', 'GPIO18'}), pull='', row=6, col=2, interfaces=frozenset({'', 'spi', 'pcm', 'dpi', 'pwm', 'gpio'}))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/python_projects/servotest.py", line 4, in <module>
    led = LED(18)
  File "/home/pi/python_projects/myproject1/lib/python3.11/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super().__call__(*args, **kwargs)
  File "/home/pi/python_projects/myproject1/lib/python3.11/site-packages/gpiozero/output_devices.py", line 192, in __init__
    super().__init__(pin, active_high=active_high,
  File "/home/pi/python_projects/myproject1/lib/python3.11/site-packages/gpiozero/output_devices.py", line 74, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/home/pi/python_projects/myproject1/lib/python3.11/site-packages/gpiozero/mixins.py", line 75, in __init__
    super().__init__(*args, **kwargs)
  File "/home/pi/python_projects/myproject1/lib/python3.11/site-packages/gpiozero/devices.py", line 553, in __init__
    pin = self.pin_factory.pin(pin)
  File "/home/pi/python_projects/myproject1/lib/python3.11/site-packages/gpiozero/pins/pi.py", line 413, in pin
    pin = self.pin_class(self, info)
  File "/home/pi/python_projects/myproject1/lib/python3.11/site-packages/gpiozero/pins/rpigpio.py", line 101, in __init__
    GPIO.setup(self._number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])
RuntimeError: Cannot determine SOC peripheral base address
```
Please advice me on how to fix this.

lurch, 2024-09-21:
> Pi model: Pi 5
Pin factory used: RPiGPIO

RPi.GPIO doesn't support the Raspberry Pi 5.

> Please advice me on how to fix this.

It should work if you switch to the lgpio pin factory https://gpiozero.readthedocs.io/en/stable/api_pins.html#changing-the-pin-factory

dshaurya, 2024-09-22:
Thank you so much. I will definitely try your method.

dshaurya, 2024-09-22:
It doesn't work.....
Terminal Code
```
GPIOZERO_PIN_FACTORY=native
```
Python code
```
from gpiozero import Device
Device._default_pin_factory()
```
The Error
```
Traceback (most recent call last):
  File "/home/pi/python_projects/myproject1/lib/python3.11/site-packages/gpiozero/pins/native.py", line 155, in __init__
    self.fd = os.open('/dev/gpiomem', os.O_RDWR | os.O_SYNC)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/dev/gpiomem'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/python_projects/myproject1/lib/python3.11/site-packages/gpiozero/pins/native.py", line 158, in __init__
    self.fd = os.open('/dev/mem', os.O_RDWR | os.O_SYNC)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/mem'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/pi/python_projects/myproject1/lib/python3.11/site-packages/gpiozero/devices.py", line 314, in _default_pin_factory
    return ep.load()()
           ^^^^^^^^^^^
  File "/home/pi/python_projects/myproject1/lib/python3.11/site-packages/gpiozero/pins/native.py", line 395, in __init__
    self.mem = GPIOMemory(self.board_info.soc)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pi/python_projects/myproject1/lib/python3.11/site-packages/gpiozero/pins/native.py", line 160, in __init__
    raise IOError(
OSError: unable to open /dev/gpiomem or /dev/mem; upgrade your kernel or run as root

```
I have updated my OS. Yet the error occurs.
I need this for a project ... Help me ASAP

lurch, 2024-09-22:
The `native` pin factory won't work on the Pi 5 either. Try using the `lgpio` pin factory like I already suggested :slightly_smiling_face:

mellivora24, 2024-11-12:
It work for me! Thanks a lot.

> sudo apt remove python3-rpi.gpio pip3 install rpi-lgpi

navyukth, 2025-03-09:
hey i ma trying us pir motion senor on rasberry pi 5 

 $ python
Python 3.11.2 (main, Nov 30 2024, 21:22:50) [GCC 12.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gpiozero import MotionSensor
>>> pir = MotionSensor(4)
Traceback (most recent call last):
  File "/home/Avyu/pi-rfid/env/lib/python3.11/site-packages/gpiozero/pins/native.py", line 155, in __init__
    self.fd = os.open('/dev/gpiomem', os.O_RDWR | os.O_SYNC)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
FileNotFoundError: [Errno 2] No such file or directory: '/dev/gpiomem'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/Avyu/pi-rfid/env/lib/python3.11/site-packages/gpiozero/pins/native.py", line 158, in __init__
    self.fd = os.open('/dev/mem', os.O_RDWR | os.O_SYNC)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
PermissionError: [Errno 13] Permission denied: '/dev/mem'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/Avyu/pi-rfid/env/lib/python3.11/site-packages/gpiozero/devices.py", line 108, in __call__
    self = super().__call__(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Avyu/pi-rfid/env/lib/python3.11/site-packages/gpiozero/input_devices.py", line 588, in __init__
    super().__init__(
  File "/home/Avyu/pi-rfid/env/lib/python3.11/site-packages/gpiozero/input_devices.py", line 257, in __init__
    super().__init__(
  File "/home/Avyu/pi-rfid/env/lib/python3.11/site-packages/gpiozero/mixins.py", line 243, in __init__
    super().__init__(*args, **kwargs)
  File "/home/Avyu/pi-rfid/env/lib/python3.11/site-packages/gpiozero/input_devices.py", line 79, in __init__
    super().__init__(pin, pin_factory=pin_factory)
  File "/home/Avyu/pi-rfid/env/lib/python3.11/site-packages/gpiozero/devices.py", line 544, in __init__
    super().__init__(pin_factory=pin_factory)
  File "/home/Avyu/pi-rfid/env/lib/python3.11/site-packages/gpiozero/devices.py", line 245, in __init__
    Device.ensure_pin_factory()
  File "/home/Avyu/pi-rfid/env/lib/python3.11/site-packages/gpiozero/devices.py", line 270, in ensure_pin_factory
    Device.pin_factory = Device._default_pin_factory()
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Avyu/pi-rfid/env/lib/python3.11/site-packages/gpiozero/devices.py", line 314, in _default_pin_factory
    return ep.load()()
           ^^^^^^^^^^^
  File "/home/Avyu/pi-rfid/env/lib/python3.11/site-packages/gpiozero/pins/native.py", line 395, in __init__
    self.mem = GPIOMemory(self.board_info.soc)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/Avyu/pi-rfid/env/lib/python3.11/site-packages/gpiozero/pins/native.py", line 160, in __init__
    raise IOError(
OSError: unable to open /dev/gpiomem or /dev/mem; upgrade your kernel or run as root
>>> 

i am keep getting this error pls help me out