gpiozero/gpiozero #1173: Monitor state of output pin

craigerl, 2024-10-07:
It is impossible for a third party program to monitor the state of an output pin.

Use case:

Program A sets output pin 12 high.
Program B cannot observe the state of pin 12 without changing it.

In my case a C program is used to key a transmitter with a gpio pin.  Meanwhile, a python program watches the output pin so it can draw a red circle on the display, indicating a transmission in progress.   This was easy with /sys/class/gpio, as you could hang an inotify on the event, but it is completely impossible with gpiozero (and libgpiod).

lurch, 2024-10-07:
I think this is "just the way that libgpiod works". Could you use some other form of IPC between your C program and your Python program? I guess in the worst-case-scenario, you could also connect the output from pin-12 to a separate GPIO input-pin, and monitor this second pin in your Python program? :thinking:

craigerl, 2024-10-07:
Yah, that's what I feared.  It will be easier for me to maintain an old kernel with the working /sys/class/gpio/*/event interface, vs teaching a thousand users to solder or make jumpers, just be cause libgpio can't monitor the state of an output pin.  Thanks for the insight!  If you ever stumble on a way to monitor a gpio pin output with gpiozero/libgpiod, please update this thread.   I'll stick with the deprecated-but-no-working-alternative /sys/class/gpio filesystem.   I can't use gpiozero/libgpiod.

lurch, 2024-10-07:
If you're willing to get your hands dirty, you _might_ be able to do what you want by adapting the source-code of https://github.com/raspberrypi/utils/tree/master/pinctrl ; but that's "going behind the kernel's back" and isn't really the way that you're "supposed" to do things.