gpiozero/gpiozero #1183: Why is setuptools a run-time requirement for gpiozero?

johndunderhill, 2024-12-17:
Operating system: Raspbian Buster
Python version: e.g. 3.7.3
Pi model: Pi 4 Model B
GPIO Zero version: gpiozero-2.0
Pin factory used: e.g. RPiGPIO

Why is setuptools a run-time requirement for gpiozero?  On my dev system (not the above hardware), this forces setuptools to install:

```
(venv) /workspace/sensors (influxdb-dev) $ python3 -m pip install -r ./requirements.txt --disable-pip-version-check
...lots of installation messages...
Collecting setuptools (from colorzero->gpiozero->-r ./requirements.txt (line 17))
  Downloading setuptools-75.6.0-py3-none-any.whl.metadata (6.7 kB)
...more installation messages...
```

Of course it won't run there unless we remote or mock the pins.  On the test system (identified above), setuptools is already present, but still.  It may not be present in our production environment (it's not a production requirement).

```
(venv) pi@raspberrypi4:~/sensors $ python3 -m pip install -r ./requirements-embedded.txt --disable-pip-version-check
...lots of installation messages...
Collecting gpiozero (from -r ./requirements-embedded.txt (line 13))
  Downloading https://www.piwheels.org/simple/gpiozero/gpiozero-2.0-py3-none-any.whl (150 kB)
...
Collecting colorzero (from gpiozero->-r ./requirements-embedded.txt (line 13))
  Downloading https://www.piwheels.org/simple/colorzero/colorzero-2.0-py2.py3-none-any.whl (26 kB)
...
Requirement already satisfied: setuptools in ./venv/lib/python3.7/site-packages (from colorzero->gpiozero->-r ./requirements-embedded.txt (line 13)) (40.8.0)
...more installation messages...
```

bennuttall, 2024-12-17:
It's covered here: https://gpiozero.readthedocs.io/en/stable/faq.html#how-can-i-tell-what-version-of-gpiozero-i-have-installed

lurch, 2024-12-17:
`pkg_resources` _used_ to be needed at runtime as part of the pin-library loading stuff, but it _looks_ like since https://github.com/gpiozero/gpiozero/commit/c45e8611fd468f87797fda8054289551fa6408a6 (which switched things to using `importlib` instead) perhaps `pkg_resources` (and therefore also `setuptools`) aren't actually needed at runtime? :thinking: