gpiozero/gpiozero #1188: Unable to set  IOPins to INPUT

hbeil, 2025-02-21:
This one baffles me.

I am using a Waveshare ESP32-S3 5inch Capacitive Touch Display Development Board, and created my Layout by utilizing their demo file(s)
The Layouts are utilizing the ESP32_Display_Panel,ESP32_IO_Expander, and the lvgl libraries. I am using 09_lvgl_Porting, if someone is familiar with these files.

The driver for IO ports and pins is the ESP_IOExpander_CH422G - this driver does not allow the state set per pin, but only per Pin group.
In the given demo file the sd card, two lcd ports and the touch screen were assigned IO pins 1-4  and explicitly set to output. 

I reassigned these pins  to  11-14  and left them alone, then reassign the standard IO pins  0-7 to INPUT. 
This does not result in the desired change. 

The debugger comes with this information:
**Thread Warning: watch: eval. expression 'enableAllIO_Input();' with no thread context. Using default** 
Default is Pins set to OUTPUT , which later checks  confirm.

Here my code from the .ino file:

#include <Arduino.h>
#include <ESP_Panel_Library.h>
#include <ESP_IOExpander_Library.h>

#include <lvgl.h>
#include "lvgl_port_v8.h"
//#include <demos/lv_demos.h>
#include <ui.h>
#include <RPM-IO.h>

// Extend IO Pin define
#define DI0 0                     // Digital Input 0
#define DI0_mask 1ULL<<DI0       // Mask for Digital Input 0
#define DI1 5                     // Digital Input 1
#define DI1_mask 1ULL<<DI1       // Mask for Digital Input 1
#define DI2 2 
#define DI3 3 
#define DI4 4 
#define DI5 1 
#define DI6 6 
#define DI7 7 
#define TP_RST 11
#define LCD_BL 12
#define LCD_RST 13
#define SD_CS 14

/**
 * To use the built-in examples and demos of LVGL uncomment the includes below respectively.
 */
 // #include <demos/lv_demos.h>
// #include <examples/lv_examples.h>

void setup()
{
    String title = "LVGL porting example";

    Serial.begin(115200);
    pinMode(GPIO_INPUT_IO_14, OUTPUT);
    /**
     * These development boards require the use of an IO expander to configure the screen,
     * so it needs to be initialized in advance and registered with the panel for use.
     */
    Serial.println("Initialize IO expander");
    
    /* Initialize IO expander */
    ESP_IOExpander_CH422G *expander = new ESP_IOExpander_CH422G((i2c_port_t)I2C_MASTER_NUM, ESP_IO_EXPANDER_I2C_CH422G_ADDRESS, I2C_MASTER_SCL_IO, I2C_MASTER_SDA_IO);
    expander->init();
    expander->begin();

    Serial.println("Set the IO0-7 pin to Input mode");
    expander->enableAllIO_Input();

    expander->digitalWrite(TP_RST , HIGH);
    expander->digitalWrite(LCD_RST , HIGH);
    expander->digitalWrite(LCD_BL , HIGH);
    delay(100);
    
    // GT911 initialization, must be added, otherwise the touch screen will not be recognized  
    // Initialization begin
    expander->digitalWrite(TP_RST , LOW);
    delay(100);
    digitalWrite(GPIO_INPUT_IO_14, LOW);
    delay(100);
    expander->digitalWrite(TP_RST , HIGH);
    delay(200);
    // Initialization end
.....

The board works fine , touchscreen works, , regardless of my changes, yet I cannot get the IOPins set to INPUT, hence I am unable to feed data to my later functions. Has anyone experienced such behavior, found a solution for the problem? Help , hints , advice is greatly appreciated.

Hartmut

closed 2025-02-21 by bennuttall

hbeil, 2025-02-21:
OK this should be request for the ESP_IOExpander_CH422G driver.

bennuttall, 2025-02-21:
This is a Raspberry Pi GPIO library, not an Arduino library