gpiozero/gpiozero #1195: Incorrect Pin Header Mapping for Raspberry Pi Compute Module 4S Rev 1.0

IrynaSemenovych, 2025-05-15:
**Environment:**

- Operating System: Raspbian Bookworm
- Python Version: 3.11
- Raspberry Pi Model: Compute Module 4S Rev 1.0
- GPIO Zero Version: 2.0.1
- Pin Factory Used: LGPIOFactory

**Issue:**
Attempting to initialize GPIO 41 with gpiozero raises a PinInvalidPin error:
```
from gpiozero import LED
led = LED(41)
```

**Error Traceback:**
```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  ...
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 416, in pin
    raise PinInvalidPin(f'{name} is not a valid pin name')
gpiozero.exc.PinInvalidPin: 41 is not a valid pin name
```

**Analysis:**
The pin_factory.board_info.headers incorrectly returns 'J8' (standard 40-pin header) as the board header.
However, the Compute Module 4S uses a SODIMM edge connector layout, not the J8 header.
This leads to GPIO pins (like 41) not being recognized, as they are not mapped in the default J8 header layout.

**Expected Behavior:**
gpiozero should detect and use the correct header layout (e.g. SODIMM) for the Compute Module 4S Rev 1.0, allowing access to GPIOs such as 41.

**Suggested Fix:**
Detect the Compute Module 4S model and assign the correct pin header layout in the board_info.headers data returned by the pin factory.

**Additional Info:**
Please see the attached full output log for detailed context.

[CM4s_gpiozero_board_info.txt](https://github.com/user-attachments/files/20233125/CM4s_gpiozero_board_info.txt)

lurch, 2025-05-18:
#1200 should fix this 🙂 
I don't have a CM4S with which to test, but I think that https://github.com/gpiozero/gpiozero/pull/1200/commits/624c81770a3f3c7e68c77a9b70c908c1754b8f12 is the minimal set of changes you actually need to get your example code working.

IrynaSemenovych, 2025-05-20:
@lurch I verified the fix in PR #1200 on CM4s. The `PinInvalidPin` exception is no longer occurring, so the fix appears to be working. Thank you for the quick response 🙂

commit 624c8177 (2025-05-21): Add data for CM4S (hopefully fixes #1195 )
closed 2025-05-21 by bennuttall