gpiozero/gpiozero #1198: list board revisions in `pinout --help`

petaflot, 2025-05-17:
current `pinout --help`
```
usage: pinout [-h] [--version] [-r REVISION] [-c] [-m] [-x]

A utility for querying GPIO pin-out information.

options:
  -h, --help            show this help message and exit
  --version             show program's version number and exit
  -r REVISION, --revision REVISION
                        Board revision. Default is to autodetect revision of current device
  -c, --color           Force colored output (by default, the output will include ANSI color codes if run in
                        a color-capable terminal). See also --monochrome
  -m, --monochrome      Force monochrome output. See also --color
  -x, --xyz             Open pinout.xyz in the default web browser
```
desired output [snip]:
```
  -r REVISION, --revision REVISION
                        Board revision. Default is to autodetect revision of current device ; one of:
                        A,B,A+,B+,2B,CM,3B,Zero,CM3,Zero W,3B+,3A+,CM3+,4B,Zero2W,400,CM4,5B
```
this is especially useful if not using autodetect

lurch, 2025-05-17:
I'm afraid that's not how the `--revision` flag works - it expects an actual hexadecimal revision-code string, as you'd find in the output of `cat /proc/cpuinfo` (e.g. `pinout -r 000e` for an early [Raspberry Pi Model 1 B](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#old-style-revision-codes) or `pinout -r c03130` for a [Raspberry Pi 400](https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#new-style-revision-codes-in-use)).

But perhaps a `--board` option might be useful?
```
  -b BOARD, --board REVISION
                        Ignore revision code and display information for the requested board; one of:
                        A,B,A+,B+,2B,CM,3B,Zero,CM3,Zero W,3B+,3A+,CM3+,4B,Zero2W,400,CM4,5B
```
(some tweaking would need to be made to the `pinout` output in this mode though, because it wouldn't be able to display the "Revision" or "RAM" fields)

lurch, 2025-05-27:
> $ pinout -r 0010
> pinout: error: 'RUN'

Yup, I've already submitted a fix for that https://github.com/gpiozero/gpiozero/pull/1200/commits/798b8e6c80943040f37c69fbbf692ba0af75b702 which got merged in #1200