gpiozero/gpiozero #1203: Raspberry Pi5 with 2 ESC engines == dead

UANEMESIS, 2025-06-08:
Operating system: Raspberry PI OS (64 bit) 2024-11-19 (not sure about name of version, but date of OS release is correct)
Python version: 3.11 as I remember
Pi model: Pi 5 
GPIO Zero version: 26.1. Release 2.0.1 (2024-02-15)

Greetings!
Dear develorers, below code that killed two my raspberry pi 5.
It sounds like a magic, but reality.
First, I thought that it was not because of script execution and raspberry died, because of other reason.
I changed dead plate to new and re-executed script and got the same result: OS freezes, after raspberry reboot, there is nothing, OS won't start and I see red indicator. Tried everything, but 2 plates is DEAD.

Logic of below code is pretty simple, I'm trying to execute scrpipt that sync with my two ESC engines and next script will start rotation.
**Important note:** 
I did the same, but using othe library on Raspberry pi 4 and it works fine. On Pi5 I lauched only one engine on PIN 12 and it works fine, but 2 PINs execution killing plate..

```
from gpiozero import PWMOutputDevice
import time

PIN_1 = 12  # RIGHT WING engine
PIN_2 = 13  # LEFT WING engine

engine_right = PWMOutputDevice(PIN_1, frequency=50)
engine_left = PWMOutputDevice(PIN_2, frequency=50)

# sync action and exit
engine_right.value = 0.03; engine_left.value = 0.03

time.sleep(4) # time for ESC beep/full sync

exit(1)
```

UANEMESIS, 2025-06-09:
@lurch ESC -> **An electronic speed control (ESC)** is an electronic circuit that controls and regulates the speed of an electric motor.
Here it is, schema (on photo PI4, because as I mentioned above, PI5 is dead):

[image]

**On PI4 it works good.**

**Regarding you links**, that's why I mentioned ESC, because basic schema (for 1 engine, but I have 2) looks like this and it's different from schemas that shown in readthedocs.

[image]

**Regarding your words:**
"...but if you push too much current or voltage through a GPIO pin it's possible to cause physical damage to the silicon chips on a Raspberry Pi, and I guess in the most extreme case this might stop the Raspberry Pi from booting?"

That's the paradocs and shock for me. Why?
1. ESC chips is not cheap and I used them before a lot on PI5(for 1 engine) and PI4 (for 2 engines)
2. ESC chips have very good voltage protection.
3. In schema with ESC + engine (my case) Raspberry pins it's about PWM command transfer. (From code we are selecting freequency and pins -- that's it!). Voltage wire is red, rest 2 wires is data (PWM) and ground. I'm not using voltage wire from ESC that can damage Raspberry:

[image]

4. All above working perfectly on PI4, but killing PI5. 💥

lurch, 2025-06-09:
I'm afraid that you're going beyond my very basic electronics knowledge, so all I can suggest is that you try asking for help at https://forums.raspberrypi.com/

UANEMESIS, 2025-06-09:
@lurch it's better, than nothing. Thanks for answer, I'm closing ticket here.

closed 2025-06-09 by UANEMESIS

lurch, 2025-06-09:
I've just noticed that in your first photo above you seem to have red, black and brown wires going directly from the Pi's GPIO header directly to the black and silver motor?? It certainly seems possible that that could cause excess currents which could overload an damage the Pi's PMIC circuitry.

UANEMESIS, 2025-06-09:
> I've just noticed that in your first photo above you seem to have red, black and brown wires going from the Pi's GPIO header directly to the black and silver motor?? It certainly seems possible that that could cause excess currents which could overload and damage the Pi's PMIC circuitry.

@lurch wires from Pi going to ESC, not directly to engine. I attached schema in my fist comment.

Moreover, first photo it's example with PI4 and working perfectly, as mentioned few times :)
I can't provide example fix PI5 (connections is the same anyway), because 2 of my PI5 is dead.

UANEMESIS, 2025-06-09:
@lurch if you are interesing, then I can record video for you, but share only i private messages or via email. Can't public it here.